ethz-asl/data-driven-dynamics #188: Online Sys ID

brunopinto900, 2022-02-21:
Hello all,

This repository allows to estimate a quadrotor's parameters based on logs. Are there any plans to perform online system identification? Think about the following examples: 

1) Attach a payload;
2) Fly against strong wind.

The idea is to estimate the parameters online and tune the controllers automatically.

Thank you.

manumerous, 2022-02-21:
Hi Bruno, 

That is a great question!

Currently we have been focusing on laying the ground work to estimate the parameters offline which allows the user to get a base dynamics model before starting a flight mission. 

But as you point out there can be cases where those parameters could change during the operation of the UAV. But I don't think that the current software infrastructure with the python package for parametric model estimation in `Tools/parametric_model`  lends itself very well to be adapted to run on an embedded system like the Pixhawk. But I don't see any reason that would inhibit us to adapt a lot of the presented approaches to run on the embedded system or to run the python code on an onboard computer like an RPi. 

Quadrotors generally use model free PID approaches on the lower controller levels. Therefore that functionality can be performed independent of the dynamics estimation in this repository. For some more advanced model based controllers (e.g. MPC) this could indeed be interesting, but the tuning there is usually specific to the chosen approach. 

You are welcome to share your ideas or plans for any project regarding online SysID and we are happy to answer any further questions about the work in this repository. 

Best, 
Manuel

manumerous, 2022-02-21:
Also regarding autotuning in the context of PX4 you can have a look at the documentation here: 
[http://docs.px4.io/master/en/config/autotune.html](http://docs.px4.io/master/en/config/autotune.html)

brunopinto900, 2022-02-21:
Thank you Manuel for the link and your insight. The auto-tune would be mainly for MPC or any other trajectory tracking controller ,for example geometric controller on @Jaeyoung-Lim repository.

manumerous, 2022-02-21:
That sounds like a really nice idea and it would of course make us very happy to see our work being able to contribute to such an effort! I am happy to answer any questions provide some support regarding such a project. But from my side I think I don't have the time to work proactively towards online SysID. 

I am sure @Jaeyoung-Lim will ad his perspective here later on.

Jaeyoung-Lim, 2022-02-22:
@brunopinto900 I believe @manumerous have already explained it quite well, but adding a few cents.

This project processes data in batches from a log file, and therefore cannot be used recursively online at its current state.

However it would definitely be interesting to compare how recursive data processing effects the accuracy of the estimated results or even if we can use them in a complementary way. (sysID input shaping depending on the recursive parameter estimates). 

On the autotuning aspect, it would definitely interesting if the model created by this repo can nicely scale into model based controller schemes.

manumerous, 2022-04-22:
I will close this issue for now, feel free to reopen it if you have additional questions.

closed 2022-04-22 by manumerous