ethz-asl/data-driven-dynamics #190: Pipeline fails with default models

Jaeyoung-Lim, 2022-02-25:
**Problem Description**
The pipeline fails with the default models

```
$ make estimate-model model=quadrotor_model
python3 Tools/parametric_model/generate_parametric_model.py \
--config /home/jaeyoung/dev/data-driven-dynamics/Tools/parametric_model/configs/quadrotor_model.yaml \
--data_selection False \
--plot True \
/home/jaeyoung/dev/data-driven-dynamics/resources/quadrotor_model.ulg
Visual Data selection enabled:  False
===============================================================================
                              Data Processing                                  
===============================================================================
Initializing of configuration successful. 
Resample frequency:  250.0 Hz
Loading uLog file:  /home/jaeyoung/dev/data-driven-dynamics/resources/quadrotor_model.ulg
Loading topics:
actuator_outputs
vehicle_local_position
vehicle_attitude
vehicle_angular_velocity
sensor_combined
Starting data resampling of topic types:  dict_keys(['actuator_outputs', 'vehicle_local_position', 'vehicle_attitude', 'vehicle_angular_velocity', 'sensor_combined'])
Resampling |################################| 6/5
Resampling |################################| 7/5Initializing of configuration successful. 
-------------------------------------------------------------------------------
Initialized dataframe with the following columns: 
['timestamp', 'u0', 'u1', 'u2', 'u3', 'vx', 'vy', 'vz', 'q0', 'q1', 'q2', 'q3', 'ang_vel_x', 'ang_vel_y', 'ang_vel_z', 'ang_acc_b_x', 'ang_acc_b_y', 'ang_acc_b_z', 'acc_b_x', 'acc_b_y', 'acc_b_z']
Data contains  14775 timestamps.
Computing force features for rotor: front right rotor
Computing moment features for rotor: front right rotor
Computing force features for rotor: back left rotor
Computing moment features for rotor: back left rotor
Computing force features for rotor: front left rotor
Computing moment features for rotor: front left rotor
Computing force features for rotor: back right rotor
Computing moment features for rotor: back right rotor
Cramer-Rao Bounds for force parameters:
vertical_rot_drag_lin    0.012782946245478006
vertical_rot_thrust_lin          0.02641080802919156
vertical_rot_thrust_quad         0.011493031809214254
c_d_fuselage_x   0.013983358667215963
c_d_fuselage_y   0.019151501301183028
c_d_fuselage_z   0.09617539023560966
Cramer-Rao Bounds for moment parameters:
vertical_c_m_leaver_quad         0.6704594437261402
vertical_c_m_leaver_lin          0.43084225232492834
vertical_c_m_drag_z_quad         0.12862700319155235
vertical_c_m_drag_z_lin          0.911563984500324
vertical_c_m_rolling     1.1950003635205884
Int64Index([ 7126,  7127,  7128,  7129,  7130,  7131,  7132,  7133,  7134,
             7135,
            ...
            12326, 12327, 12328, 12329, 12330, 12331, 12332, 12333, 12334,
            12335],
           dtype='int64', length=1973)
===============================================================================
                        Preparing Model Features                               
===============================================================================
===============================================================================
                            Initialize Optimizer                               
                                QPOptimizer
===============================================================================
Define and solve problem:
min_c (X * c -y)^T * (X * c -y)
 s.t. G * c <= h
Initialized with the following coefficients: 
['vertical_rot_drag_lin', 'vertical_rot_thrust_lin', 'vertical_rot_thrust_quad', 'vertical_c_m_leaver_quad', 'vertical_c_m_leaver_lin', 'vertical_c_m_drag_z_quad', 'vertical_c_m_drag_z_lin', 'vertical_c_m_rolling', 'c_d_fuselage_x', 'c_d_fuselage_y', 'c_d_fuselage_z']
Fixed Coefficients: Value
-------------------------------------------------------------------------------
Bounded Coefficients: (Min Value, Max Value)
vertical_rot_drag_lin:  (0.0, 2.0)
vertical_rot_thrust_lin:  (-5.0, 0.0)
vertical_rot_thrust_quad:  (0.0, 50.0)
vertical_c_m_leaver_quad:  (0.0, 2.0)
vertical_c_m_leaver_lin:  (-1.0, 0.0)
vertical_c_m_drag_z_quad:  (0.0, 2.0)
vertical_c_m_drag_z_lin:  (-1.0, 0.0)
vertical_c_m_rolling:  (0.0, 2.0)
c_d_fuselage_x:  (0.0, 2.0)
c_d_fuselage_y:  (0.0, 2.0)
c_d_fuselage_z:  (0.0, 2.0)
[ 1.48865838e-01 -3.84856814e-02  4.10355783e+00  5.33476003e-01
 -2.93407944e-02  4.83691240e-01 -4.77612430e-04  7.12698257e-01
  2.22379266e-02  2.09342308e-02  1.86787335e-01]
===============================================================================
                           Optimization Results                                
===============================================================================
                           Optimal Coefficients                              
-------------------------------------------------------------------------------
c_d_fuselage_x: 0.022237926620199577
c_d_fuselage_y: 0.020934230791888102
c_d_fuselage_z: 0.18678733457238925
vertical_c_m_drag_z_lin: -0.00047761242954867683
vertical_c_m_drag_z_quad: 0.4836912404668401
vertical_c_m_leaver_lin: -0.029340794412230996
vertical_c_m_leaver_quad: 0.5334760026177074
vertical_c_m_rolling: 0.7126982570997248
vertical_rot_drag_lin: 0.14886583777313
vertical_rot_thrust_lin: -0.03848568139202761
vertical_rot_thrust_quad: 4.103557829442402

-------------------------------------------------------------------------------
                            Optimization Metrics                               
-------------------------------------------------------------------------------
R2: 0.9998113181856075
RMSE: 0.0760681581901144

-------------------------------------------------------------------------------
Complete results saved to: 
model_results/multirotor_model_2022-02-25-15-01-36.yaml
-------------------------------------------------------------------------------
Traceback (most recent call last):
  File "Tools/parametric_model/generate_parametric_model.py", line 146, in <module>
    start_model_estimation(**vars(arg_list))
  File "Tools/parametric_model/generate_parametric_model.py", line 128, in start_model_estimation
    model.compute_residuals()
  File "/home/jaeyoung/dev/data-driven-dynamics/Tools/parametric_model/src/models/dynamics_model.py", line 486, in compute_residuals
    y_forces_pred[0::3] = y_pred[0:int(self.y_forces.shape[0]/3)]
ValueError: could not broadcast input array from shape (11838) into shape (14775)
Makefile:34: recipe for target 'estimate-model' failed
make: *** [estimate-model] Error 1
```

pascalau, 2022-02-28:
I think this was caused due to the accidentally enabled automatic data selection and was not caught by the unit tests because of the disabled plotting https://github.com/ethz-asl/data-driven-dynamics/blob/ab3189de38dbd5db62bc3111e76e4b8e16150ce4/.github/workflows/sysid_test.yml#L32 .

Jaeyoung-Lim, 2022-02-28:
Fixed in https://github.com/ethz-asl/data-driven-dynamics/pull/191

closed 2022-02-28 by Jaeyoung-Lim