ethz-asl/data-driven-dynamics #198: Simulation and testing problem (inconsistent behavior) and can't test the functionality of the parametric mode

AEGEAN99, 2022-04-22:
Hello,

I am having a problem with simulating the model on Gazebo. When I try to take off the drone take-offs successfully but soon it starts to oscillate then gets out of control and crashes. I tried both my own log files and the example default log file supplied from the data-driven-dynamics repo. I am sending command take-off from qgroundcontrol. 
And my

manumerous, 2022-04-22:
Thank you for reporting this issue. :) We are happy to help you and to further improve the pipeline. 
For this we might need some more information. Could you share the full results file you obtain when using the default log? it should be located in the `model_results` folder.

AEGEAN99, 2022-04-22:
[quadrotor_model.zip](https://github.com/ethz-asl/data-driven-dynamics/files/8540990/quadrotor_model.zip)
I build the data-driven-dynamics to PX4-Autopilot/build/px4_sitl_default/data-driven-dynamics

AEGEAN99, 2022-04-22:
Should I open any different log file setting from pixhawk parameters on my quad? Or standart log files are enough?

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Could you also share the full log on your terminal?

AEGEAN99, 2022-04-22:
of course  here my termina  when training:
`make estimate-model model=quadrotor_model log=resources/quadrotor_model.ulg
python3 Tools/parametric_model/generate_parametric_model.py \
--config /home/a3g34n/PX4-Autopilot/build/px4_sitl_default/data-driven-dynamics-master/Tools/parametric_model/configs/quadrotor_model.yaml \
--data_selection False \
--plot True \
resources/quadrotor_model.ulg
Visual Data selection enabled:  False
===============================================================================
                              Data Processing                                  
===============================================================================
Initializing of configuration successful. 
Resample frequency:  250.0 Hz
Loading uLog file:  resources/quadrotor_model.ulg
Loading topics:
actuator_outputs
vehicle_local_position
vehicle_attitude
vehicle_angular_velocity
sensor_combined
Starting data resampling of topic types:  dict_keys(['actuator_outputs', 'vehicle_local_position', 'vehicle_attitude', 'vehicle_angular_velocity', 'sensor_combined'])
Resampling |################################| 6/5
Resampling |################################| 7/5Initializing of configuration successful. 
-------------------------------------------------------------------------------
Initialized dataframe with the following columns: 
['timestamp', 'u0', 'u1', 'u2', 'u3', 'vx', 'vy', 'vz', 'q0', 'q1', 'q2', 'q3', 'ang_vel_x', 'ang_vel_y', 'ang_vel_z', 'ang_acc_b_x', 'ang_acc_b_y', 'ang_acc_b_z', 'acc_b_x', 'acc_b_y', 'acc_b_z']
Data contains  14775 timestamps.
Computing force features for rotor: front right rotor
Computing moment features for rotor: front right rotor
Computing force features for rotor: back left rotor
Computing moment features for rotor: back left rotor
Computing force features for rotor: front left rotor
Computing moment features for rotor: front left rotor
Computing force features for rotor: back right rotor
Computing moment features for rotor: back right rotor
Cramer-Rao Bounds for force parameters:
vertical_rot_drag_lin 	 0.012782946245478023
vertical_rot_thrust_lin 	 0.02641080802919156
vertical_rot_thrust_quad 	 0.011493031809214254
c_d_fuselage_x 	 0.013983358667215987
c_d_fuselage_y 	 0.01915150130118305
c_d_fuselage_z 	 0.09617539023560966
Cramer-Rao Bounds for moment parameters:
vertical_c_m_leaver_quad 	 0.6704594437261402
vertical_c_m_leaver_lin 	 0.43084225232492834
vertical_c_m_drag_z_quad 	 0.12862700319155235
vertical_c_m_drag_z_lin 	 0.911563984500324
vertical_c_m_rolling 	 1.1950003635205881
===============================================================================
                        Preparing Model Features                               
===============================================================================
===============================================================================
                            Initialize Optimizer                               
                                QPOptimizer
===============================================================================
Define and solve problem:
min_c (X * c -y)^T * (X * c -y)
 s.t. G * c <= h
Initialized with the following coefficients: 
['vertical_rot_drag_lin', 'vertical_rot_thrust_lin', 'vertical_rot_thrust_quad', 'vertical_c_m_leaver_quad', 'vertical_c_m_leaver_lin', 'vertical_c_m_drag_z_quad', 'vertical_c_m_drag_z_lin', 'vertical_c_m_rolling', 'c_d_fuselage_x', 'c_d_fuselage_y', 'c_d_fuselage_z']
Fixed Coefficients: Value
-------------------------------------------------------------------------------
Bounded Coefficients: (Min Value, Max Value)
vertical_rot_drag_lin:  (0.0, 2.0)
vertical_rot_thrust_lin:  (-5.0, 0.0)
vertical_rot_thrust_quad:  (0.0, 50.0)
vertical_c_m_leaver_quad:  (0.0, 2.0)
vertical_c_m_leaver_lin:  (-1.0, 0.0)
vertical_c_m_drag_z_quad:  (0.0, 2.0)
vertical_c_m_drag_z_lin:  (-1.0, 0.0)
vertical_c_m_rolling:  (0.0, 2.0)
c_d_fuselage_x:  (0.0, 2.0)
c_d_fuselage_y:  (0.0, 2.0)
c_d_fuselage_z:  (0.0, 2.0)
[ 1.59066672e-01 -1.38275472e-01  4.19292061e+00  3.39425013e-01
  1.88560297e-20  5.07582893e-01 -3.12479704e-02  7.30058131e-22
  2.08259718e-03  1.08326852e-02 -6.22803436e-13]
===============================================================================
                           Optimization Results                                
===============================================================================
                           Optimal Coefficients                              
-------------------------------------------------------------------------------
c_d_fuselage_x: 0.002082597182204064
c_d_fuselage_y: 0.010832685190091253
c_d_fuselage_z: -6.228034359331524e-13
vertical_c_m_drag_z_lin: -0.03124797041333541
vertical_c_m_drag_z_quad: 0.5075828926443368
vertical_c_m_leaver_lin: 1.885602973253947e-20
vertical_c_m_leaver_quad: 0.33942501322805907
vertical_c_m_rolling: 7.300581311503944e-22
vertical_rot_drag_lin: 0.15906667162903768
vertical_rot_thrust_lin: -0.13827547174291024
vertical_rot_thrust_quad: 4.19292060688978

-------------------------------------------------------------------------------
                            Optimization Metrics                               
-------------------------------------------------------------------------------
R2: 0.9876949185774294
RMSE: 0.609447037495574

-------------------------------------------------------------------------------
Complete results saved to: 
model_results/multirotor_model_2022-04-22-16-07-16.yaml
`

AEGEAN99, 2022-04-22:
and figures
[image]
[image]
[image]
[image]
:

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Could you try with the "LinearRegressor" Instead of the "QPOptimiser"?

also might help if you updte to latest master

AEGEAN99, 2022-04-22:
Actually I dont know how to change optimizer. Should I download the latest master PX4-Autopilot? Or should I install the version in your repo?

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Sorry I modified my initial response.

AEGEAN99, 2022-04-22:
okay I am trying now

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 The plots remind me of a problem that I fixed a while ago: https://github.com/ethz-asl/data-driven-dynamics/pull/193 Probably worth checking if the version you are using has the changes included

AEGEAN99, 2022-04-22:
still not working,
Here is latest one with qpoptimizer
[image]
[image]
[image]
[image]
[image]

AEGEAN99, 2022-04-22:
And here latest with linearRegressor
[image]
[image]
[image]
[image]
[image]

AEGEAN99, 2022-04-22:
This happens much more slower

https://user-images.githubusercontent.com/62438751/164732107-130c77c6-11ac-4598-8abe-d5402b182809.mp4

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Ah, now it is working. 

Could you try 
```
make estimate-model model=quadrotor_model log=resources/quadrotor_model_trajectory.csv
```

AEGEAN99, 2022-04-22:
with linearRegressor?

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Yes, it is basically just swapping the log

AEGEAN99, 2022-04-22:
It is working but has small oscillations like vibratio

https://user-images.githubusercontent.com/62438751/164733933-a0de9032-6568-45f8-b60c-f0d51dc600f4.mp4

n

AEGEAN99, 2022-04-22:
Here is the figures of the model. Are they satisfying?
And have you determined the problem?
[image]
[image]
[image]
[image]
[image]

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Yes, so this means that it has successfully estimated the dynamics.

I am not sure why the vehicle is oscillating, but as you have probably seen, it is important that you have information in your logs in order to make the parameters visible.

AEGEAN99, 2022-04-22:
But when I try the quadrotor_model.csv it agin crashes

AEGEAN99, 2022-04-22:
What should I do for my own log files

Jaeyoung-Lim, 2022-04-22:
@AEGEAN99 Yes, because it does not contain enough information - as you can see in the logs there is no angular excitation of the system and the system is unable to successfully estimate the correct parameters

Same for your own logs - You need a log that has enough information to estimate the parameters.

manumerous, 2022-04-22:
Thanks for taking over Jay! 

Yes I agree with you. I will quickly make a PR and put the trajectory log as our default log. That is indeed confusing. 

Furthermore I think this issue points towards the fact that we need to improve the user feedback on the estimated parameters. 

@AEGEAN99 as Jay already pointed out estimating the angular dynamics is usually harder than the linear ones. Can you try to fly some maneuvers where you have a lot of angular acceleration and try the pipeline again?

manumerous, 2022-04-22:
Adressed in #199

AEGEAN99, 2022-04-22:
Thank you so much!! I will try flying the drone next weeks and update this comment. I hope that works. Thank you for your all effort))))))

manumerous, 2022-04-23:
Thanks for reporting the issues!

closed 2022-04-24 by AEGEAN99