ethz-asl/data-driven-dynamics #221: Unified formatter for python code

manumerous, 2023-04-13:
In order to have a unified style and avoid white space changes it would be good to have a unified formater. Should be use Black or do you have a special preference?

Jaeyoung-Lim, 2023-04-13:
@manumerous Yes please, I was actually looking for one for python

manumerous, 2023-04-13:
I did not do a very detailed evaluation, but Black seems to be a good option. It is the most popular formatter at the moment and actively enforces all code to be compliant with pep8 and its own rules. So I would suggest to move forward with that. I guess the rest is preference anyways.

sjschlapbach, 2023-04-16:
Sounds like a great idea to me! I started out with some different formatters, but black seems to be a nice solution to have a well-integrated approach for all Python files in the repository.
I took the liberty to already format all files with it and integrate the formatting rule into the style checking with PR #222.

closed 2023-04-16 by sjschlapbach