ethz-asl/data-driven-dynamics #223: Model estimation broken for upstream PX4

Jaeyoung-Lim, 2023-04-16:
**Problem Description**
The model estimation pipeline is broken for upstream PX4

To reproduce:
```
 make estimate-model model=vtol_tiltrotor_model log=resources/2832a4ac-2906-4327-a5f0-98db8df6e3e3.ulg 
python3 Tools/parametric_model/generate_parametric_model.py \
--config /home/jaeyoung/dev/data-driven-dynamics/Tools/parametric_model/configs/vtol_tiltrotor_model.yaml \
--data_selection none \
--plot True \
resources/2832a4ac-2906-4327-a5f0-98db8df6e3e3.ulg
===============================================================================
                              Data Processing                                  
===============================================================================
Initializing of configuration successful. 
Resample frequency:  50.0 Hz
Loading uLog file:  resources/2832a4ac-2906-4327-a5f0-98db8df6e3e3.ulg
Loading topics:
actuator_outputs
vehicle_local_position
vehicle_attitude
vehicle_angular_velocity
sensor_combined
manual_control_setpoint
vehicle_thrust_setpoint
vehicle_torque_setpoint
vehicle_land_detected

Computing flight time...
More than one flight detected. Please check the landed state.
Starting data resampling of topic types:  dict_keys(['actuator_outputs', 'vehicle_local_position', 'vehicle_attitude', 'vehicle_angular_velocity', 'sensor_combined', 'manual_control_setpoint', 'vehicle_thrust_setpoint', 'vehicle_torque_setpoint', 'vehicle_land_detected'])
Resampling |################################| 9/9
Initializing of configuration successful. 
-------------------------------------------------------------------------------
Initialized dataframe with the following columns: 
['timestamp', 'u0', 'u1', 'u2', 'u3', 'vx', 'vy', 'vz', 'q0', 'q1', 'q2', 'q3', 'ang_vel_x', 'ang_vel_y', 'ang_vel_z', 'ang_acc_b_x', 'ang_acc_b_y', 'ang_acc_b_z', 'acc_b_x', 'acc_b_y', 'acc_b_z', 'aux1', 'aux2', 'aux3', 'aux4', 'aux5', 'aux6', 'throttle', 'aileron', 'elevator', 'rudder', 'landed']
Data contains  0 timestamps.
Computing force features for rotor
Computing moment features for rotor
Traceback (most recent call last):
  File "Tools/parametric_model/generate_parametric_model.py", line 252, in <module>
    start_model_estimation(**vars(arg_list))
  File "Tools/parametric_model/generate_parametric_model.py", line 171, in start_model_estimation
    model.prepare_regression_matrices()
  File "/home/jaeyoung/dev/data-driven-dynamics/Tools/parametric_model/src/models/dynamics_model.py", line 106, in prepare_regression_matrices
    self.prepare_force_regression_matrices()
  File "/home/jaeyoung/dev/data-driven-dynamics/Tools/parametric_model/src/models/fixedwing_model.py", line 101, in prepare_force_regression_matrices
    ) = self.aero_model.compute_aero_force_features(
  File "/home/jaeyoung/dev/data-driven-dynamics/Tools/parametric_model/src/models/aerodynamic_models/linear_wing_model.py", line 160, in compute_aero_force_features
    v_airspeed_mat[0, :], angle_of_attack_vec[0], elevator_input_vec[0]
IndexError: index 0 is out of bounds for axis 0 with size 0
make: *** [Makefile:34: estimate-model] Error 1
```

sjschlapbach, 2023-04-16:
@Jaeyoung-Lim Is the corresponding ulog-File available somewhere online to reproduce the issue?

Jaeyoung-Lim, 2023-04-16:
@sjschlapbach That was where I was just at :) 

`num_of_groups` is 4 which looks like a case not handled in the code

sjschlapbach, 2023-04-16:
@Jaeyoung-Lim Thanks, now this makes total sense... I didn't take a look at the `landed` topic yet, but if I draw the right conclusions from these logged messages, the log contains "two flights" in some sense where the second one probably occurred from a bump during landing?
[image]

What's the behavior that would be expected? To just use the first section where the landed topic is 1 or to concatenate all parts to allow for flights with multiple parts?

Jaeyoung-Lim, 2023-04-16:
I think we want to take all the data while the vehicle is "in air"...but good question.

I am not sure if land detector would be reliable enough to assume it only triggers during flight. But certainly good idea to take the data out when the land detector has triggered.

sjschlapbach, 2023-04-16:
Alright, this makes sense. I can have a look at this in the coming days. I think it should be fairly easy to fix this with the test log you provided.

sjschlapbach, 2023-04-27:
@Jaeyoung-Lim I addressed this issue with PR #225, which also worked when I tested it with the flight log you provided. I hope this solves the problem :)

closed 2023-04-27 by Jaeyoung-Lim