ethz-asl/data-driven-dynamics #224: ID data gathering FW/VTOL

sfuhrer, 2023-04-17:
As promised last week, I did data gathering flights on 3 different fixed-wing types, each including the same type of maneuvers, and each containing 3 ID phases, marked with  `manual_control_setpoint.aux1`:
1. throttle 0, pitch sinusoidal
2. pitch 0, throttle sinusoidal
3. combined throttle and pitch inputs
(4. the ZMO flight contains a 4th phase, with also combined pitch/throttle inputs).

Further, each flight contains prolonged (at least a couple of minutes) of steady loiter flight, as well as steady climb/sinking phases that can be used for validation of the model. The ZMO flight additionally contains flying at a higher airspeed setpoint.

Solius FW:
https://review.px4.io/plot_app?log=a967d7a5-10fa-41ca-b43a-1abaebce044a
[image]
[image]

Loong standard VTOL:
https://review.px4.io/plot_app?log=ec92ddd8-46d4-4847-9cb0-8e2a81dddeb4
[image]
[image]

ZMO Tiltrotor VTOL:
https://review.px4.io/plot_app?log=b2ecbb01-9a67-40c5-8846-b46f0ce04d8e
[image]
[image]

Note: it was a quite sunny day, and on other planes we experienced quite strong uplifts due to thermals. This could well cause inconsistencies in the measured angle of attack, as to have that observable we currently assume 0 vertical wind.

Jaeyoung-Lim, 2023-04-17:
Awesome!

sfuhrer, 2023-04-17:
@Jaeyoung-Lim  @manumerous @sjschlapbach @tstastny  Let's use this flight data to:
- find out if for each vehicle the model can be identified simply with this one flight
- if the non-Id part of the flights can be used for validation
- if the separated maneuvers (pitch or throttle) are yielding superior results to the combined excitation (note: I noticed that I in general flew too fast during the combined excitation phases, so they may be invalid due to that)
- generate TECS tuning values for each plane and compare them to the current controller tuning
- validate if the assumption of 0 vertical wind is causing mayor inaccuracies

manumerous, 2023-04-18:
Thanks a lot! That looks great!