ethz-asl/data-driven-dynamics #233: self.prepare_regression_matrices() issue.

anheinle, 2023-11-16:
I have a question regarding the prepare_regression_matrices(self): function. When called: 

`X, y = self.prepare_regression_matrices()` it expects two variables (X, y) in the return, but the function seems to be returning nothing. Could that be a bug? 

The error I get is the following: 

```
Traceback (most recent call last):
  File "Tools/parametric_model/predict_model.py", line 108, in <module>
    start_model_prediction(**vars(arg_list))
  File "Tools/parametric_model/predict_model.py", line 70, in start_model_prediction
    model.predict_model(opt_coefs_dict)
  File "/home/anna/Workspaces/ddd_ws/src/data-driven-dynamics/Tools/parametric_model/src/models/dynamics_model.py", line 491, in predict_model
    X, y = self.prepare_regression_matrices()
TypeError: cannot unpack non-iterable NoneType object
make: *** [Makefile:43: predict-model] Error 1
```

Thank you in advance!

sjschlapbach, 2023-11-16:
Hi @anheinle,

thank you for your message! Could you provide some more information so that we can try to reproduce the difficulties you are encountering, as for example:
- Which model do you use for the parameter identification?
- Could you provide us a log file, which does result in this error message?
- Could you provide us the exact command you used to run the pipeline?

The failing function you mention is used in all constellations, and works in the cases we have recently tested.

anheinle, 2023-11-17:
1. First I ran `make estimate-model model=quadrotor_model log=resources/quadrotor_model.ulg` and used the saved model for further processing (model_results/multirotor_model_2023-11-17-07-56-56.yaml)
2. I have tried the command `make predict-model` with different log files including your log file (resources/quadrotor_model.ulg):
`make predict-model model=quadrotor_model log=resources/quadrotor_model.ulg model_results=model_results/multirotor_model_2023-11-17-07-56-56.yaml data_selection=False`

sjschlapbach, 2023-11-22:
Dear @anheinle,

I think I managed to identify the issue in the meantime and implement a corresponding fix (PR #234). As soon as this is merged, the estimation and prediction with the commands you posted should work again.

If you want to test the solution already, feel free to check out the corresponding branch. I hope this resolves your problems with the pipeline.

closed 2023-11-23 by Jaeyoung-Lim