ethz-asl/data-driven-dynamics #237: Are real flight logs available

Azircarry, 2024-04-22:
Hello, I have collected some flight data of my real aircraft (quadcopter UAV), can I use the model recognition function directly? Need some other action? I'm sorry to bother you.

Jaeyoung-Lim, 2024-04-22:
@Azircarry Did you experience any problems when trying to process your data?

Azircarry, 2024-04-22:
> 您在尝试处理数据时是否遇到任何问题？

Sorry, I haven't actually done the identification of the real data yet. I just wanted to ask first if I needed additional operations or if I could just use the feature. (I intend to use ulg file format for data)

Azircarry, 2024-04-23:
> 您在尝试处理数据时是否遇到任何问题？

Hello, when I run the make command, I get the following error (I really can't find CommandMotorSpeed.pb.h), what is the problem
[image]

Azircarry, 2024-04-23:
> 您在尝试处理数据时是否遇到任何问题？

Does this plugin need to be installed in the PX4-Autopilot directory?I'm sorry to bother you.I was hoping you could help me

Azircarry, 2024-04-24:
> @Azircarry Did you experience any problems when trying to process your data?

Hello, I downloaded four log files in     https://logs.px4.io/browse      and when I ran it I got the following (one ran successfully, the other three got an error)
[image]
[image]
[image]
[image]
It seems like it's all  actuator_outputs  problem, I'm not sure.Can you help me? Thanks

Azircarry, 2024-04-24:
> > 您在尝试处理数据时是否遇到任何问题？
> 
> 您好，我在 https://logs.px4.io/browse 下载了四个日志文件，当我运行它时，我得到了以下内容（一个成功运行，其他三个出现错误） ![4bda6d775320302618f0f2d0b7f804c](https://private-user-images.githubusercontent.com/114214508/325066566-e4de2bb4-bd6b-4fe3-af0e-6931e654b1d4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5Mjk3MTIsIm5iZiI6MTcxMzkyOTQxMiwicGF0aCI6Ii8xMTQyMTQ1MDgvMzI1MDY2NTY2LWU0ZGUyYmI0LWJkNmItNGZlMy1hZjBlLTY5MzFlNjU0YjFkNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNFQwMzMwMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNGU0ZmVmMmFkOGJhNmE5MjVjMGE4ZTAzMjk2OTE0MTMxZmY0MzQ0OGRkZTU2NmQ2OGM4Yjg0MDQ1YWNjZDlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DiI1NJAozF7opOfmkC3L7JrIe3neIENQOP-0BiynV1c) ![ca222490ce98c0e3998f18f2dc00cdf](https://private-user-images.githubusercontent.com/114214508/325066607-1199ab99-ef56-4977-b081-ffa093ee3c23.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5Mjk3MTIsIm5iZiI6MTcxMzkyOTQxMiwicGF0aCI6Ii8xMTQyMTQ1MDgvMzI1MDY2NjA3LTExOTlhYjk5LWVmNTYtNDk3Ny1iMDgxLWZmYTA5M2VlM2MyMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNFQwMzMwMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYmViMTk2NTIzZmUyNGFhYTYxMzZjMDE3MzYzNDZkMzMwYjkwMzMwN2VmYTI4NTQ3ODBhYTI5ZTgwNmE5MGRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JebwomX8WeKSJuDrpKWB1SRQu-H_XbLKuC6OGngvXAo) ![3e0dc7a362b706373ec0b5827e4bb4d](https://private-user-images.githubusercontent.com/114214508/325066638-c102f6c5-3264-4ebb-a421-93faaabf8f27.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5Mjk3MTIsIm5iZiI6MTcxMzkyOTQxMiwicGF0aCI6Ii8xMTQyMTQ1MDgvMzI1MDY2NjM4LWMxMDJmNmM1LTMyNjQtNGViYi1hNDIxLTkzZmFhYWJmOGYyNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNFQwMzMwMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNTYyZmExMGIyNTE2ZjBlOGRjZTM2ODZmYjM0MmFiMzc0MzFmMDE5NTY0NmM1Y2ZmZGY1NWMzMmE4MDM5Y2JlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XJurTrbptDqE6sRKt9CokoXTyZkEHaqXByiNBmDu1oM) ![927fe589ebff8db3278d54b0c0642e2](https://private-user-images.githubusercontent.com/114214508/325066677-28409e00-ca03-453d-add1-74da90ea4825.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5Mjk3MTIsIm5iZiI6MTcxMzkyOTQxMiwicGF0aCI6Ii8xMTQyMTQ1MDgvMzI1MDY2Njc3LTI4NDA5ZTAwLWNhMDMtNDUzZC1hZGQxLTc0ZGE5MGVhNDgyNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNFQwMzMwMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wM2IzZmJhNzI2ZDIyMDRhNTk1NmQxNGY1YzY3ZDI1MTBmNDk0M2U0YmRlZWZmOGM5YjI3NjAxNjc3YzI2Y2IxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EuGEguy17qvzqj4xC4YU9d7lP04Wwhnl-7WC_z8KrBI)似乎这都是actuator_outputs问题，我不确定。你可以帮我吗？谢谢

This is the flight log archive I used to test
[Uploading QUADTEST.zip…]()

Azircarry, 2024-04-24:
[QUADTEST.zip](https://github.com/ethz-asl/data-driven-dynamics/files/15087504/QUADTEST.zip)

Jaeyoung-Lim, 2024-06-28:
@Azircarry Name of uORB topics can change depending on the px4 version. I suggest you try the flight logs in https://github.com/ethz-asl/data-driven-dynamics/issues/224