ethz-asl/data-driven-dynamics #241: Unable to Run the Simulation

MohamedAhsan97, 2024-09-17:
[image]

When I try to run the simulation with the commands provided in the Readme file, I get this error (screenshot attached). 
Please advise me how to run the simulation. 

I am working with PX4 Vision autonomy drone, and managed to generate the .yaml file for the quadrotor_model.ulg using the plugin.

manumerous, 2024-09-17:
Hi Mohamed, did you follow the instructions step by step in the Readme to setup the environment. In particular the commands ```cd <Firmware Dir>
DONT_RUN=1 make px4_sitl gazebo```

and `` echo "export PX4_ROOT=~/src/PX4-Autopilot" >> ~/.bashrc
source ~/.bashrc```

If yes could you post the results you see in the terminal so we can check whether the build succeded?

MohamedAhsan97, 2024-09-20:
Hello. Thank you for your response. 

Yes, I did follow the instructions in the Readme file. The PX4-Autopilot plugin builds fine with the commands mentioned, and the data-driven plugin also builds successfully. But when I run these commands to run the simulation 

source setup.bash
Tools/sitl_run.sh -m iris -s iris_aerodynamics

I am getting this error;
[image]

This is my terminal output when PX4 is built;
[image]
[image]

And these are my outputs when i build the data-driven plugin;
[image]

manumerous, 2024-09-20:
The error seems to be that a file in `PX4-Autopilot/Tools/sitl_gazebo/worlds/empty.world` can not be found. I assume you validated that the file does not exist? 

I took a quick look into the PX4 repository on github and it seems like on the latest commit there is no more `sitl_gazebo` folder present in Tools.
[https://github.com/PX4/PX4-Autopilot/tree/main/Tools](https://github.com/PX4/PX4-Autopilot/tree/main/Tools)
In that case they likely changed something without us adapting the data-driven-dynamics pipeline yet. @Jaeyoung-Lim do you know by chance where that config might have moved?

Jaeyoung-Lim, 2024-09-21:
@manumerous Sorry for the late response. The path has now moved to `PX4-Autopilot/tree/main/Tools/simulation/gazebo-classic/sitl_gazebo-classic/worlds/empty.world`